mbarszcz-pcb/c64-rf-modulator-replacement #3: GND on long boards.

bwack, 2020-07-25:
Hi, I'm helping a guy on facebook regarding ground on long boards. He noticed that the GND is not mentioned in the readme (under jumper settings). I'm just presuming from the schematics, that the G.PAL jumper on the mother board must be joined/solder-blobbed for it to work ?
Looks neat btw !! :)

modeler, 2020-07-25:
I am that guy, and thank you bwack for helping out. With the original modulator in place, the far-left of the eight pins (as you look at the top of the board) is connected to ground via the modulator housing. When removed, the pin is not connected to ground unless the G.PAL jumper is shorted as bwack says.

Thank you for this project!

mbarszcz-pcb, 2020-07-25:
I didn't see the issue on Facebook and never noticed before that pin "8" (gnd on the long boards) is only gnd when the G.PAL jumper is closed.

Looking at the notes in the original commodore schematic, it seems that the jumper's original function would have been to change the sound carrier in the original modulator.  Since no audio is passing through the RF modulator replacement anymore (or being modulated onto an RF signal), I guess the easiest thing would just be to make sure that jumper is closed when using the RF Modulator.  Was that the solution you came to as well?

bwack, 2020-07-26:
Yes!

closed 2020-09-30 by bwack