mbarszcz-pcb/c64-rf-modulator-replacement #4: UK PAL 250407 boards have floating ground pin to the modulator

davepoo, 2021-04-10:
Not really an issue but making you aware that UK PAL 250407 boards have floating ground pin to the modulator. So your modulator replacement actually requires the solder jumper on the board to be connected to make the ground good, or it will be loaded with interference. Might be worth adding a note about this to the readme.

https://youtu.be/jXSQe5f-Rsk?t=428

I made a video on this while i was installing the modulator replacement (the bad ram and other fails on the computer are not related this ground problem)

davepoo, 2021-04-10:
Oh and accoring to the internet. https://www.cavsi.com/questionsanswers/what-are-the-pal-variants/
This "I Pal" with the open ground will affect "UK, Ireland, Hong Kong and Macau" versions of the boards.

There are five varieties of PAL:

1. PAL B/G/D/K/I
The majority of countries using PAL have television standards with 625 lines and 25 frames, differences concern the audio carrier frequency and channel bandwidths. Standards B/G are used in most of Western Europe, standard I in the UK, Ireland, Hong Kong and Macau, standards D/K in most of Eastern Europe and Standard D in mainland China.

7-MHz channels are used in VHF (B, D) and 8-MHz channels in UHF (G, K, I), although Australia used 7-MHz channels in UHF and Ireland uses 8-MHz channels in VHF .

davepoo, 2021-04-10:
Actually, i've just noticed that there was already a closed issue for this
https://github.com/mbarszcz-pcb/c64-rf-modulator-replacement/issues/3

Too late, i had to figure it out myself.

mbarszcz-pcb, 2021-04-10:
Thanks for bringing this to my attention (again).  I will add a section to the main page regarding this to hopefully eliminate confusion in the future.  None of my boards had that jumper open so it never came up for me.  The ground pin was just...ground.