mbarszcz-pcb/c64-rf-modulator-replacement #6: Could this work on the C16 as well?

SukkoPera, 2021-11-09:
Hi, thanks for your project. I was wondering if this circuit could work in the Commodore 16, too. I am asking because I would like to integrate it in [a clone mainboard I am making](https://github.com/SukkoPera/LittleSixteen) (with the obvious modifications for power, input signals, etc).

Here are the schematics of the C16/Plus4 modulator, taken from [the Service Manual](https://github.com/SukkoPera/LittleSixteen/blob/master/doc/C16_Service_Manual_314001-03_(1984_Oct).pdf).

[image]

Your schematics look similar but not exactly the same. Do you think I could have a go with your circuit, which is a bit simpler and more tailorable (I like the pots!) or should I use the exact circuit from the above?

Thanks for your time! :)

SukkoPera, 2021-11-14:
Same schematics as above with RF stuff removed. The similarity with this project is now more evident.

[image]

SukkoPera, 2021-11-14:
I have modified the schematics of this project adding the missing components and noting the different values of other components:
[image]

Now, the 3 added caps and the added inductor are also present, with similar values, in the [schematics of the 251696 modulator for the C64](http://www.zimmers.net/anonftp/pub/cbm/schematics/computers/c64/modulator-251696.gif), which makes me think they aren't critical, if they were omitted from this project. This is also stated explicitly in [this other C64 modulator replacement circuit I've found](https://i2.wp.com/oms.wmhost.com/misc/modulator_simple.png?zoom=2).

The rest of the circuit is exactly the same as this project's, with only a few different values. I don't have the knowledge to judge whether the different values are critical or not, but I think I can dare say that **the circuit of this project will probably work on the C16 as well**, possibly with a few different values for some components. I guess I'll go with it for the prototype of my boards. Once they are manufactured I will do some testing and report back.

Note that the current board of this project will NOT be a drop-in replacement for the C16, as the connecting pins are placed differently, some redesign will be needed.

mbarszcz-pcb, 2021-11-22:
I'm sorry I am not familiar with the C16 enough to be able to judge whether it will work or not.

SukkoPera, 2021-12-22:
No problem, we went ahead and ordered the boards. We'll soon find out :).